ZeroPhone/Discussion-and-Research #13: VO-WiFi??

JustinFuhrmeister-Clarke, 2018-05-09:
I dont know, but how hard would it be to implement vowifi/volte as the Phone function.

and use the gsm system for data only?

CRImier, 2018-05-09:
It seems that having VoLTE is going to be easy - seeing as it's the default implementation of voicecall functionality on LTE networks (so we can't really avoid it as we'll need call capability), and at least some of the Simcom LTE modems seem to support it. Thus, once a 4G (LTE) modem hardware is available (just like 3G hardware now is), we should be able to just use VoLTE calling functionality the same way we use 2G/3G phone calls.

VoWiFi, on the other hand, is a different story - much like, say, eSIM, it's a technology that doesn't yet seem to be available on modems like the ones we're using (and I'm not sure it will be any soon). It's a cool technology, but I'm not sure we'll get suitable hardware any soon =(

JustinFuhrmeister-Clarke, 2018-05-09:
I didn't realise that volte and vowifi were hardware based, i though that they were or could be implemented by software.

But yes possibly a solution later on.

Keep up the awesome work